dathlin/HslCommunication #1: demo 已经下载不下来了 404了  删除了？

pencilso, 2019-08-28:
https://github.com/dathlin/HslCommunication/raw/master/Download/HslCommunicationDemo.zip

pencilso, 2019-08-28:
订阅模式在哪订阅啊  大兄弟？

dathlin, 2019-08-28:
demo已经上传了，项目已经闭源了，参考：
http://www.hslcommunication.cn/Cooperation

pencilso, 2019-08-28:
对于收费   我没有意见，公司也同意   但是，目前我看不到源码，我不太确定是否能符合我们的需求。
我们想做到    类似于Http服务器 ，大概是   Android 、 iOS 通过类似于Http的协议 远程调用HslCommunication  然后再由HslCommunication  与设备进行通讯，不知能否实现？

pencilso, 2019-08-28:
公司商用，可以不需要HslCommunication源码，需要Java调用HslCommunication的代码，是不是普通VIP群就够了

dathlin, 2019-08-29:
你的需求肯定是可以实现的，这个毋庸置疑，普通vip群是个人用户的，你可以先试试看，目前C#版本的.dll在nuget是免费下载和使用的。

closed 2019-08-29 by pencilso

alanchiang0210, 2020-05-29:
请问
1. http://www.hslcommunication.cn/Cooperation 这个网开不来
2. 目前是否个人与企业用户都需要付费

dathlin, 2020-05-29:
参照DEMO的的界面：

| |
Chasing
|
|
hsl200909@163.com
|
签名由网易邮箱大师定制

请问

http://www.hslcommunication.cn/Cooperation 这个网开不来
目前是否个人与企业用户都需要付费

alanchiang0210, 2020-05-29:
我有个朋友已经用了5.7版很久了, 可以继续使用吗, 是否需要付费呢

dathlin, 2020-05-29:
可以继续免费用，只不过建议升级