dathlin/HslCommunication #22: Any one know whether HSL can read/write AB Controllogix time?

U11Leung, 2022-04-26:
As the title says, does anybody know or ever try to read/write plc time on Allen-Bradley ControlLogix/CompactLogix? Searched the API document but did not find anything related

dathlin, 2022-04-26:
I'm very sorry, there is currently no built-in API to read and write time data, but you can read raw byte data, and you can parse it into time type data yourself.

U11Leung, 2022-04-26:
Appreciate your quick reply! But I want to know more about raw byte data? Is by reading raw byte data I can read the plc local time, which is displayed here? How am I reading that?
[image]

dathlin, 2022-04-26:
There are currently two methods to try.
1. Use DEMO to browse the node content of PLC, there is a special CIP protocol node browsing interface. Check to see if there is a system node or the like, the system node may have been blocked, and the success rate of this method is not high.
2. If your PLC program can access the time variable, assign the variable data to the common node, and then access the common node variable.

U11Leung, 2022-04-27:
Sorry, but I did not see anything like that you said?
[image]

[image]