dathlin/HslCommunication #27: Hsl Communication调试工具或java调取接口批量读取三菱、AB等协议报远程关闭问题

sjdGitHub, 2022-07-01:
当我单个读取时没有什么问题，但是实际上使用大部分的场景都是批量读取点位数据，批量读取时Hsl Communication调试工具会报“Read failed:错误代号:-1 文本描述:Socket Exception ->远程关闭了连接”，代码调用也是单个读取不报错，批量读会报远程关闭了连接这样的错误。不知道这是bug还是其他问题？或者说代码有更新？

dathlin, 2022-07-01:
使用了什么类测试的，连接了什么PLC？型号是什么？进行了什么批量操作？

sjdGitHub, 2022-07-01:
感谢您的回答，我调试的时候先用了Hsl Communication调试工具分别创建了三菱的MC virtual server、AB的MC virtual server，作为虚拟服务器，然后根据API文档分别调用MelsecMcNet.ReadRandom(String[] address, short[] length)、AllenBradleyNet.Read(String[] address, int[] length)进行批量读取点位数据。发现都不行，后面就用调试工具里的批量读取测试发现也不行。具体如下：
[image]
[image]
[image]

dathlin, 2022-07-01:
这个原因是虚拟服务器并未实现批量读取功能，这部分的指令比较复杂，后续有时间实现
你可以直接连接真实的PLC测试，如果还有问题，可以联系我

sjdGitHub, 2022-07-01:
好的，非常感谢您的回复