dathlin/HslCommunication #38: requesting to contact support

hugodavidx, 2023-03-17:
why It is asking for commercial authorization??

Connection NOK！Error Code: :10000
Description: :System authorization failed, need to use activation code authorization, thank you for your support. Active device number：2910494 If you need commercial authorization, please contact Email: xxxxxxx@xxx.com

dathlin, 2023-03-17:
Hello, very sorry, this library needs to be activated to be used for a long time, otherwise it can only be used for 24 hours, the specific commercial authorization can be emailed to: hsl200909@163.com

hugodavidx, 2023-03-17:
Why did not you post that before implement your library?, so, that mean I need to pay for that authorization?

dathlin, 2023-03-17:
I'm very sorry, we provide 24 hours of testing to better allow users to test the library more, and it is indicated on GitHub that this is a paid library.

hugodavidx, 2023-03-17:
there are not any specification about the 24 hours, you should add at some place to be easier to communicate it. this looks like fraudulent