dathlin/HslCommunication #39: 西门子字符串 读到历史值怎么处理

nm-1216, 2023-04-14:
西门子字符串 读到历史值怎么处理

dathlin, 2023-04-15:
不太明白，读到历史值是什么意思

nm-1216, 2023-04-17:
[image]

nm-1216, 2023-04-17:
怎么拿实际值？ 有方法么

nm-1216, 2023-04-17:
[image]

随便写了一个，不知道hsl 里面有没有自带

dathlin, 2023-04-17:
在hsl里有自带的，例如你得字符串存储在 DB1.0  里面，那么可以直接调用下面的方法

`
OperateResult<string> read = plc.ReadString("DB1.0");
if (read.IsSuccess)
{
     string value = read.Content;
}
`

nm-1216, 2023-04-17:
不写长度就是得？

nm-1216, 2023-04-17:
> 在hsl里有自带的，例如你得字符串存储在 DB1.0 里面，那么可以直接调用下面的方法
> 
> `OperateResult<string> read = plc.ReadString("DB1.0"); if (read.IsSuccess) { string value = read.Content; }`

不行，必须输入长度，西门子PLC，开始两个字符是 总长度与有效长度，得转

nm-1216, 2023-04-17:
[image]
[image]

dathlin, 2023-04-17:
最新版不需要输入，内部自动分析字符串长度的，或是传0也可以。

如果你用的特别老的版本，则可能不行

closed 2023-04-18 by nm-1216