dathlin/HslCommunication #45: GE SRTP仿真软件

lylyuanliang, 2023-06-21:
请问对于GE SRTP协议的验证, 有什么软件可以作为参考呢,

dathlin, 2023-08-01:
hslcommunicationdemo本身提供了这个虚拟服务器，可以用作没有PLC时候的测试

lylyuanliang, 2023-08-11:
我使用hsl虚拟模拟器, 然后发现了个问题, 暂时不知道什么原因造成的

**前置描述:**
内存类型: R
操作步骤: R1写入long类型数据9223372036854775807, R9写入short类型数据32767

**问题描述1**
按照我的推测, long类型数据占用了8个字节, R1为long类型, 那么我的下一个内存地址得为R9才不会影响R1的值, 但实际测试发现,  我对R5写入值都不会影响R1的值, (对R4写入会影响), 
我的疑问是: R1已经为long了, 为啥下一个可用的地址是R5而不是R9呢

dathlin, 2023-08-11:
因为R1 一个地址占2个字节

lylyuanliang, 2023-08-11:
> 因为R1 一个地址占2个字节

基于这个, 现在发现了另一个问题,  R1写入字符串 abcdef, 长度为6, 按照一个地址2个字节, 我在R4写入应该不会有问题, 但实际R4写入后, R1的值受到了影响
[image]

dathlin, 2023-08-11:
你截图中字符串读取，前面有个长度，默认是10，你要读取R1-R3的字符串的话，长度输入 3 就可以

lylyuanliang, 2023-08-11:
> 你截图中字符串读取，前面有个长度，默认是10，你要读取R1-R3的字符串的话，长度输入 3 就可以

感谢大佬! 另外再额外请教一下, GE SRTP设计的内存地址都是一个地址占两个字节吗, 我看到代码中WordLength为2