dathlin/HslCommunication #48: 使用modbusTCP连接了一天左右，突然连接写不进去值了，重启之后就可以连接上继续写值

sl9411, 2023-10-08:


sl9411, 2023-10-08:
_wcld_client = new ModbusTcpNet(_ipaddr, _port);
                    OperateResult OptRet = _wcld_client.ConnectServer();
                    if (OptRet.IsSuccess)
                    {
                        Log.LogInstance.WriteLog("连接成功", LogType.Run, "Cooler" + _devno + "Log");
                        _net_connected = true;
                        StartWriteHeartBeatThread();
                        StartWcldStateThread();
                    }
                    else
                    {
                        Log.LogInstance.WriteLog("连接失败", LogType.Run, "Cooler" + _devno + "Log");
                        _net_connected = false;
                    }

sl9411, 2023-10-08:
我有多个设备需要连接，使用的都是这样一中连接方式，多台不同的设备同时写不进去值了。其他设备采用别的连接方式没用使用此类库连接正常

dathlin, 2023-10-09:
库本身如果不激活的话，只能连续使用24小时，重启软件又是24小时，并且在读写失败后，Message显示授权失败的消息。
授权参考：http://www.hsltechnology.cn/Home/Licence?area=HslCommunication

yogu2017, 2025-01-13:
这种软件不适合在开源里面出现了吧

dathlin, 2025-01-13:
> 这种软件不适合在开源里面出现了吧

我们的DEMO示例程序本身是免费开源的，只是依赖了一个收费库，库的长期发展确实需要大家的支持，感谢理解和支持